copterust/mpu9250 #6: Allow for 20MHz SPI

nraynaud, 2019-03-03:
The MPU 9250 allows for a 20MHz clock SPI while reading the value registers but only allows a 1MHz SPI when configuring.

I have a fork of this lib where I remove the spi peripheral from the MPU struct, and passed a mutable reference to it as an argument to `Mpu9250::marg()` and `Mpu9250::.all()`.

Are you interested in a PR? It needs significant cleanup, so I won't do the work if it's a non-starter.

The result is a IUM read of 11.us instead of 151us, which is enough gain that I can answer and interrupt, do a blocking read, and compute magwick at 4kHz on a STM32F103 MCU (let it be clear that there is not a lot of slack left in the timing, 35us).

I hope this logic analyser capture is not too messy to read (the 8MHz capture rate can't show the actual SPI clock): 
[image]

the principal interrupt handler is this (LED1 is the actual one from the preceding traces): 
```rust
    #[allow(non_snake_case)]
    #[interrupt(resources = [LED1, EXTI, MPU, FILTER, SPI, ], priority = 2)]
    fn EXTI15_10() {
        resources.EXTI.pr.modify(|_, w| w.pr15().set_bit());
        resources.LED1.set_high();
        let a: MargMeasurements = resources.MPU.all(resources.SPI).unwrap();
        let w = imu::V {
            x: a.gyro.x,
            y: a.gyro.y,
            z: a.gyro.z,
        };
        let a_val = imu::V {
            x: a.accel.x,
            y: a.accel.y,
            z: a.accel.z,
        };
        *resources.FILTER = imu::filter_update(w, a_val, *resources.FILTER, 1.0 / 10.0);
        resources.LED1.set_low();
    }
```

Here is a section from the MPU 9250 datasheet on the topic:
[image]

little-arhat, 2019-03-03:
Great find! 

I'm not a big fan of removing spi from mpu struct itself, as it complicates things quite a bit and make it hard to work with I2C (see recent PR to this repo).

AFAIK, embedded-hal doesn't provide any API for re-configuring SPI bus (and that's questionable, of course), so we can't handle this inside MPU as well.

Can we add additional constructor parameter `reconfiguration_fn` of type `SPI -> SPI` (or `mpu::Device -> mpu::Device` in more general case), so users can reconfigure bus after initialization?

bofh, 2019-03-03:
Or better add .at_speed() (or speed parameter to transfer?) to SPI which will be turned to noop in case the bus is already at that speed by the compiler.

nraynaud, 2019-03-03:
A few points: 
 - the SPI configuration is not in embedded-hal but in the MCU crates, they are easier to change, and I more or less expect everybody to have a clone of them locally. Another point is that I am trying to get some PRs in the MCU I use.
 - I don't like MPU having exclusive access to SPI for all its life, SPI busses are meant to be shared, that was the whole point of the NSS pin. Same reasoning for I2C.
 - We could add a function parameter for the reconfiguration, but I was thinking of switching off the SPI between the interrupts, which is another use case that needs exclusive access to the peripheral
 - changing the interface of SPI in embedded-hal is almost impossible because of the RFC process, I'd better not go there. Moreover, speed is never mentioned at this level of abstraction, that will make discussion even more weird.

little-arhat, 2019-03-03:
Regarding sharing: one way to achieve that is using smth like https://github.com/Rahix/shared-bus, has its drawbacks too, of course. Not sure it can help with switching bus off between interrupts, though. 

Yeah, changing embedded-hal traits doesn't look easy, unfortunately.

nraynaud, 2019-03-03:
Do you need an explicit mutex on the bus? RTFM seems to implement generic mutexes. 

The code I posted earlier shows that the SPI is protected by a mutex, and the read is blocking. RTFM does the task priority checks.

here is the code in context: https://github.com/nraynaud/pill-rust-test/blob/master/src/main.rs#L208

mciantyre, 2019-03-03:
If you choose to use `shared-bus`, and the RTFM framework guarantees that you don't need a mutex, could you implement a `NoOpMutex` to satisfy `shared-bus`?

```rust
use core::marker::PhantomData;
pub struct NoOpMutex<T>(PhantomData<T>);

impl<T> shared_bus::BusMutex<T> for NoOpMutex<T> {
    fn create(_: T) -> NoOpMutex<T> {
        NoOpMutex(PhantomData<T>)
    }

    fn lock<R, F: FnOnce(&T) -> R>(&self, _: F) -> R { /* do nothing */ }
}
```
I wouldn't recommend this unless RTFM really gives you that guarantee (I've never used it myself, so I can't speak to that).

little-arhat, 2019-03-03:
(linking embedded-hal discussions: https://github.com/rust-embedded/embedded-hal/issues/35 & https://github.com/eldruin/xca9548a-rs/issues/1)

nraynaud, 2019-03-03:
reading, I didn't know about those discussions.

little-arhat, 2019-03-18:
Fixed in #8, closing for now.

closed 2019-03-18 by little-arhat