copterust/mpu9250 #10: Calling mpu.calibrate_at_rest() results in blank mag readings

jamwaffles, 2019-03-16:
I have an MPU9250 that I'm using with a Madgwick filter. I set it up like this (on an STM32F103):

```rust
let ncs = gpioa.pa4.into_push_pull_output(&mut gpioa.crl);
let scl_sck = gpioa.pa5.into_alternate_push_pull(&mut gpioa.crl);
let ad0_sdo_miso = gpioa.pa6;
let sda_sdi_mosi = gpioa.pa7.into_alternate_push_pull(&mut gpioa.crl);
let mpu_spi = Spi::spi1(
    device.SPI1,
    (scl_sck, ad0_sdo_miso, sda_sdi_mosi),
    &mut afio.mapr,
    mpu9250::MODE,
    1.mhz(),
    clocks,
    &mut rcc.apb2,
);

let mut mpu = Mpu9250::marg_default(mpu_spi, ncs, &mut delay).expect("Failed to init MPU");

mpu.calibrate_at_rest(&mut delay).expect("MPU calibration failed");
```

I have a timer interrupt running at 220Hz which contains (among other things) this line:

```rust
let all: MargMeasurements = resources.MPU.all().expect("Failed to read values");
```

When I call `hprintln!("{:?}", all.mag);` to print the mag readings to the debug console, I get 

```
Matrix { data: [0.0, 0.0, 0.0] }
```

(and for some reason a NaN for the roll component from Madgwick). 

If I remove the `calibrate_at_rest()` call, the mag readings are non-zero again. I've tried adding delays between the `marg_default()` call `and `calibrate_at_rest()` to no avail. Do I need to assign the returned `Vector3()` to something? The docs aren't clear on how to use the returned value, but relevant to this issue is why the mag readings are zeroed after `calibrate_at_rest()`.

closed 2019-03-17 by little-arhat

little-arhat, 2019-03-17:
Hi @jamwaffles, thanks for report!

0.9.5 will fix this. The issue was that `ak8963` (on-board mangetrometer) wasn't re-initialized after calibration. 

I've also updated docs to make them a bit more clear, but maybe they still aren't %)

The idea of `calibrate_at_rest()` is to detect gyro & accel biases and store them to special registers, so they will be applied to readings "automatically" by sensor itself. However, setting `accel` biases didn't work, so that's why `calibrate_at_rest` returns them, so users can apply them if they wish to. 

I'm not sure, btw, this method should be used. I think calibration itself and setting biases should be decoupled, and mpu crate should provide methods for setting/retrieving biases, and maybe helper method to perform calibration. But for now, you can use `calibrate_at_rest()` and then apply returned accel biases on top of accel readings (gyro biases will be applied automatically).

Hope that helps!

little-arhat, 2019-03-17:
https://github.com/copterust/proving-ground/blob/master/semi_sensors/main.rs

jamwaffles, 2019-03-17:
That's awesome, thank you for the fix! I'll give it a test tonight when I have the hardware in front of me.

commit 7d8f5895 (2019-03-17): Re-init magnetrometer (ak8963) after calibration.

Fixes #10.

Since we reset device to perform calibration, we have to re-init it
afterwards. Previously we've been only doing that for IMU (6axis), so
magnetrometer remained un-initiaized. This commit fixes that.