copterust/mpu9250 #14: Documentation states that mag values are in tesla, is this correct?

pftbest, 2019-06-03:
Documentation states that values are in tesla:
https://github.com/copterust/mpu9250/blob/cff5b06b0793a1f315fe7a71bf662bc4538dc781/src/lib.rs#L1296

but the values I get seem too large to be in tesla. Also the example code names them as micro tesla, which seems more realistic:
https://github.com/copterust/mpu9250/blob/cff5b06b0793a1f315fe7a71bf662bc4538dc781/examples/bbblue.rs#L32

closed 2019-06-03 by little-arhat

pftbest, 2019-06-04:
I think `m` prefix [usually means milli](https://physics.nist.gov/cuu/Units/prefixes.html), not micro. For micro we use `u` or `µ`.

https://github.com/copterust/mpu9250/blob/ef970bfd78c5dff497a51788aa1b7a21e7653855/src/lib.rs#L1296

little-arhat, 2019-06-04:
@pftbest yes, of course! sorry for being stupid %( I've pushed update.

commit a6fea0f3 (2019-06-03): Fix mag units in docs, fixes #14