copterust/mpu9250 #15: Advantages/Disadvantages of I2C (vs SPI)

michael-p, 2019-06-25:
Dear all,
thanks a lot for this nice and well-designed crate!

Is there some information available regarding the advantages and disadvantages of interfacing the MPU-9250 via I2C, compared to SPI? Obviously I2c support in this crate is still experimental and hence less mature but are there fundamental differences, and do you plan to support the full feature-set with I2c as well?

(Background of my question is that I work on a project which uses a custom Arm-board where the Mpu-9250 is connected via I2c and I have to decide whether we need to change this to SPI in the next revision, or whether we can continue using I2c, which I would strongly prefer)

Thank you very much!
Michael

bofh, 2019-06-25:
It is all about your:
* pin budget (2 vs 4);
* required bus speed (SPI is “faster”);
* distance/noise on a bus;
* power budget (I do believe that i2c draws more);
* presence of other devices on a same bus;
* countless other differences between those two.

I strongly suggest you to choose the right bus based on the physical requirements for your device, not on the current state of rust driver.

Driver could be improved cheap, bad design decisions not that easy.

bofh, 2019-06-25:
Rule of thumb: SPI for high speed/low-power applications with a couple of chips, i2c for many devices on a bus/multi-master environment.

michael-p, 2019-06-26:
Hey, thanks a lot for your tipps! :)
In my particular case the electrical/physical requirements are not an issue (we already have a working board using I2C and according to our EE SPI would be fine as well) so my question really is about the software and chip capabilities. I had the impression that most people (esp. in Arduino community) use I2c for this sensor and was a bit surprised to see that this crate has only experimental support for it.

mciantyre, 2019-07-03:
I'll comment on why I consider I2C support experimental:

The crate was not designed for both SPI and I2C support. Back in the original crate [here](https://github.com/japaric/mpu9250), there was no I2C support, and there [was no intention on supporting I2C](https://github.com/japaric/mpu9250/issues/5). Given the ideas in that issue, and with some study of the crate's development, we identified a way that this crate could support both SPI and I2C. It works, but it's not perfect. One example that comes to mind: the original implementation assumed many chip I/O actions were accomplished by a SPI transfer, which leaves the zeroth byte of the transfer buffer untouched. I2C doesn't do that, so throughout the implementation we need to [ensure that I2C actions can accommodate for this expectation](https://github.com/copterust/mpu9250/blob/master/src/device.rs#L326:L337). We either allocate a buffer with an extra byte, or use some tricks that let us ignore the zeroth byte, as demonstrated in that snippet. A driver crate that was designed up-front for both SPI and I2C might not have these kinds of quirks, and it may have abstractions in place that would more easily support both busses.

The I2C implementation has gone through much less testing and usage when compared to the SPI implementation. My use-case is limited (a BeagleBone Blue w/ MPU9250 for quickly prototyping other systems). From my understanding, the crate maintainers are using an MPU9250 with a SPI bus. If I make a change, I'm not testing against an SPI-based platform to make sure I didn't break anything. Likewise, the crate maintainers may not be testing against an I2C-based platform. We do our best to coordinate changes, but sometimes [changes to support I2C break SPI](https://github.com/copterust/mpu9250/issues/12). I would consider a crate with first-class SPI and I2C support to have contributors with vested interest in each implementation. Unfortunately, given my availability and obligations, I cannot always be that champion for I2C. The maintainers have demonstrated their ability to support the SPI implementation; therefore, the SPI implementation has first-class support.

That being said, when reading sensor data, I believe I2C implementation should be on par with the SPI implementation. One thing that's not supported is communicating with the AK8963 via the MPU9250. We're using I2C pass-through, with the host directly addressing the AK8963. However, just as the SPI master talks to the magnetometer using the MPU, an I2C master could do the same.

bofh, 2019-07-04:
That's probably due to ease of wiring on a breadboard. We usually have SPI bus in flight controllers.

michael-p, 2019-07-08:
Thank you all for your valuable comments, that answers all the questions I had (and hopefully everyone else's coming across this issue later on).

closed 2019-07-08 by michael-p