copterust/mpu9250 #16: Support UART

lorenzleutgeb, 2019-07-08:
I am trying to read from a MPU9250 (as part of a JY901 module) connected to my computer (running Linux) via a USB-to-serial converter (a CP210x). The converter is detected and mounted as `/dev/ttyUSB0`. Is there any way I could get this driver working with my setup? If so, please sketch out how that should be implemented, and I will try to execute that.

mciantyre, 2019-07-08:
I do not believe this crate will support your use-case. If you were building the JY901 module itself, the `mpu9250` crate could be useful. The `mpu9250` crate requires direct communication via SPI (or I2C). Although the JY901 module is using an MPU9250, it does not seem to directly expose the sensor to the user.

As described in [the reference manual](http://wiki.wit-motion.com/english/lib/exe/fetch.php?media=module%3Awt901%3Adocs%3Ajy901usermanualv4.pdf), the device presents a UART interface. Chapter 7, section 2, seems to describe the communication and addressing scheme. Have you tried interfacing the JY901 using an existing serial tool, or an existing (Rust) library? I'm thinking a crate like [serialport](https://crates.io/crates/serialport) should work.

mciantyre, 2019-07-08:
[Here](https://github.com/dkm/python-jy901) is an example of a Python user interfacing the JY901. Notice that they're [using the Python serial library to construct a JY901 object](https://github.com/dkm/python-jy901/blob/master/conf.py#L38:L39), which is then used to read from the motion sensor according to the documented protocol. My understanding is that you're looking for something like this, but written in Rust, correct?

lorenzleutgeb, 2019-07-09:
That's what I expected. Thanks a lot for providing me with a reference to that Python script. Using it I could confirm that my module is working correctly. Now I'll try to replicate it with [`tokio-serial`](https://crates.io/crates/tokio-serial).

closed 2019-07-09 by lorenzleutgeb