copterust/mpu9250 #20: Builder Pattern for MpuConfig

tchamelot, 2020-03-25:
Currently, `MpuConfig<MODE>` tries to follow the [Builder Pattern](https://doc.rust-lang.org/1.0.0/style/ownership/builders.html). However, to fully match the pattern, `MpuConfig<MODE>` is missing a function like

```rust
impl MpuConfig<MODE> {
    pub fn consume(&self) -> Self {
        self
    }
}
```

The problem is that currently `MpuConfig<MODE>` has only default builders which return `Self` or setters which return `&mut Self`. That makes it really difficult to use. Would a refactor of the configuration be possible. My current idea would be to add:

```rust
impl MpuConfig<MODE> {
    pub fn build<Device>(&self,  dev: Device) -> Mpu9250<D, MODE> {
        Mpu9250 {
            dev,
            raw_mag_sensitivity_adjustments: [0; 3],
            mag_sensitivity_adjustments: [0.0; 3],
            gyro_scale: self.gyro_scale.unwrap_or_default(),
            accel_scale: self.accel_scale.unwrap_or_default(),
            mag_scale: self.mag_scale.unwrap_or_defaut(),
            accel_data_rate: self.accel_data_rate
                .unwrap_or_default(),
            gyro_temp_data_rate: self.gyro_temp_data_rate
                .unwrap_or_default(),
            sample_rate_divisor: self.sample_rate_divisor,
            _mode: PhantomData,
        }
    }
}
```

Actually, just that would pose a problem as the `Device` creation is now to be handled by the final user. But we could imagine having two `build` for I2C and SPI which create the `Device`.

Like for my previous issue, I am up for it.

little-arhat, 2020-03-25:
Hey, @tchamelot! I was thinking recently about something like [this](https://xaeroxe.github.io/init-struct-pattern/), though not sure if it's possible to achieve without breaking changes.

tchamelot, 2020-03-25:
Yeah, I have seen this article on reddit, I don't know if everything he uses is stable or not (neither if all that this crate use is stable too as I work with nightly) but this would be a good idea.

little-arhat, 2020-03-26:
@tchamelot btw, "Like for my previous issue, I am up for it" — sure, I agree that current approach to configuration is not ideal and would love to improve it, so feel free to take a stab (either by improving builder pattern or by using "struct init"  or maybe other approaches)! %)