copterust/mpu9250 #32: Get/set biases for both mpu and dmp

mariusknaust, 2020-06-17:
I will start to implement this. Just fyi, so no work is done twice.

Hopefully setting the accelerometer biases will work as well. If this is the case, should the calibrate_at_rest method be changed to directly set them instead of returning them? This would be a breaking change, otherwise it would be a little bit counterintuitive.

bofh, 2020-06-17:
I’m not exactly sure that having calibrate_at_rest in a driver is a good idea anyway: there are countless ways to calibrate IMU.

tchamelot, 2020-06-17:
For the dmp there is a features that enable auto calibration. It has been implemented in #26. I have tested it and I have compared it with the use of `calibrate_at_rest` in the dmp initialization. The result are the same (same measures with either method). The calibration of the accelerometer is not set, but after some test with the original librobotcontrol for the beagelbone blue, I did not notice any change with and without accelerometer calibration. (Maybe, my mpu is perfectly calibrated, we don't know the odds). However, what are you exactly planning to do?

> I’m not exactly sure that having calibrate_at_rest in driver is a good idea anyway: there are countless ways to calibrate IMU.

The mpu9250 and its family have hardware register to perform on board calibration. If we don't want to provide a calibration function for directly in the driver, we should at least make it available in the API.

mariusknaust, 2020-06-18:
I just want to expose the bias registers with get/set methods, so they can be used for any sort of calibration. My initial plan was to do that for the dmp bias memory locations as well. But, it seems like the normal mpu biases have influence on the dmp quaterion calculation too. So there is no real need for it and I might skip it.

The auto calibration of #26 does gyro calibration for the dmp only and writes the bias into the mentioned dmp bias memory location.

I will leave the calibration_at_rest method like it is for now. I agree, there are different ways to to calibration and especially for the accelerometer it is also dependent on how it is mounted. Nevertheless, it would be nice to have a simple go to method for it, like that existing function (taken from the sparkfun Arduino implementation). Especially because it makes use of the fifo, which btw might interfere with the dmp configuration in the current form?

closed 2020-08-14 by tchamelot