copterust/mpu9250 #37: How to pass mpu9250 handle to a function

liebman, 2022-10-15:
Maybe I'm missing something as my rust-foo is new.  I'm trying to pass ownership to a function. The receiving function is declared as:
```
    pub fn new<E, DEV, MODE>(mpu: Mpu9250<DEV, MODE>) -> Result<IMU, Box<dyn Error>> 
    where 
        DEV: Device<Error = E>
```
Yet when I attempt to call the `mpu.all()` I get this:
```
error[E0599]: no method named `all` found for struct `Mpu9250<DEV, MODE>` in the current scope
  --> src/imu.rs:40:63
   |
40 |                     let data: ImuMeasurements<[f32; 3]> = mpu.all()?;
   |                                                               ^^^ method not found in `Mpu9250<DEV, MODE>`
   |
   = note: the method was found for
           - `Mpu9250<DEV, Imu>`
           - `Mpu9250<DEV, Marg>`
```

Any idea on what I'm doing wrong? (I've tried many variations on the signature)

liebman, 2022-10-15:
Ok - this works - *sigh* - would be nice if there was a trait or set of traits for this:
```
    pub fn new<E, I2C, MODE>(mpu: Mpu9250<I2cDevice<I2C>, Imu>) -> Result<IMU, Box<dyn std::error::Error>> 
    where I2C: i2c::Read<Error = E>
              + i2c::Write<Error = E>
              + i2c::WriteRead<Error = E>
```
Oh - yea - I probably should have said up front I'm using I2C.

little-arhat, 2022-10-17:
@liebman "would be nice if there was a trait or set of traits for this" you mean trait for `i2c::Read<Error = E> + i2c::Write<Error = E> `+` i2c::WriteRead<Error = E>`? [embedded-hal](https://github.com/rust-embedded/embedded-hal/) is probably better place for such trait? Maybe https://github.com/rust-embedded/embedded-hal/pull/223 could be used

closed 2023-07-29 by little-arhat