copterust/mpu9250 #39: the trait bound `u16: cast::From<_>` is not satisfied

mwbryant, 2023-07-28:
Sorry if this is a beginner question but I'm just starting with Rust embedded.  I am attempting to use this crate with an arduino UNO.

When I add mpu9250 as a dependency to my toml the project fails to compile with 57 errors in the form:
```

error[E0277]: the trait bound `u16: cast::From<_>` is not satisfied
    --> /home/matthew/.cargo/registry/src/github.com-1ecc6299db9ec823/mpu9250-0.24.2/src/lib.rs:1810:44
     |
1810 |          ((u16(buffer[offset + 6]) << 8) + u16(buffer[offset + 5])) as i16]
     |                                            ^^^^^^^^^^^^^^^^^^^^^^^ the trait `cast::From<_>` is not implemented for `u16`
     |
     = help: the trait `cast::From<f64>` is implemented for `f32`

```

I am able to compile the crate when I clone it from git.

I'm following this blog: https://blog.logrocket.com/complete-guide-running-rust-arduino/ which uses the github template: https://github.com/Rahix/avr-hal-template and all I have changed from the template is adding this dependency.

little-arhat, 2023-07-29:
Error comes from [cast crate](https://crates.io/crates/cast) we use to perform safe conversions. 
`cast` itself generates functions/implementations based on the `target_pointer_width`: https://github.com/japaric/cast.rs/blob/main/src/lib.rs#L355. 

I don't know much about arduino uno, but [seems](https://internals.rust-lang.org/t/adding-16-bit-pointer-support/2484) it has 16bit pointers, so `cast` crate doesn't generate required functions %( 

We need to fix this in `mpu9250`, perhaps by not using `cast` at all, and doing conversions by hand. 

Thanks for reporting, will provide new version shortly!

little-arhat, 2023-07-29:
@mwbryant please try `0.25.0`!

closed 2023-07-29 by little-arhat

mwbryant, 2023-07-29:
Works great!  Thank you so much!

commit 55f7e6e6 (2023-07-29): Do not use cast crate, perform casts manually

Fixes #39

`cast` crate generates "safe" type conversion functions based on
`target_pointer_width`. Unfortunately, crate only handles 32 & 64 bit
widths, so it's not complete for `rust-avr`/`Arduino` with 16bit
pointers.

This commits removes `cast` crate in favour of manual `as` type casts.
`mpu9250` only does safe conversions anyway, so benefits of `cast`
crate are negligible.