copterust/mpu9250 #40: Support different I2C addresses

pkb-pmj, 2023-08-16:
MPU9250 can have one of two I2C addresses: `0x68` or `0x69`. That way you can use two MPUs on one I2C bus.
As far as I know, SPI doesn't have anything similar.
Is it possible to add this feature, given that the crate is built around SPI?

bofh, 2023-08-16:
Chip select signal is used to select device on SPI bus, so you need to have an output pin per sensor to choose to which to talk to.

bofh, 2023-08-16:
But you right, we need to add AD0 support in I2C mode.

little-arhat, 2023-08-16:
Also, fyi, it's possible to share one i2c bus using smth like [shared_bus](https://crates.io/crates/shared-bus): [example](https://github.com/copterust/proving-ground/blob/master/shared_i2c/main.rs).

pkb-pmj, 2023-08-17:
I've just realized, it would have been hard if `MPU9250` was also a trait, but since it's a `struct impl`, it can be different for I2C and SPI... and even already is.
So it's just a matter of adding `address` to `I2cDevice` and replacing `i2c` with `i2c, address` in all the "constructors", just like the existing `spi, ncs`.
I'll add a PR today.